rust-embedded-community/embedded-nal #4: self mutable reference

MathiasKoch, 2020-02-18:
I am a bit unsure on why this shouldn't have a mutable reference to self?

The way i see it, you would usually implement this on some sort of network client, owning access to a peripheral (unix fd, SPI, I2C, Serial etc.), but most (all?) of these require a mutable reference to be able to send data. This could be fixed by wrapping the peripheral in a `RefCell`, but even then there is a good chance that the client object would hold stuff like internal state etc that needs to be mutated, meaning that in my particular case, i would end up with a client object of 5 fields, all wrapped in `RefCell`.

Is this intended behavior, or am i implementing something wrong in general?

thejpster, 2020-02-20:
I had intended implemetors of the crates to handle internal mutability. While in the BSD socket library, I can call `open()` or `write()` etc, without having to hold any kind of pointer or reference - it is handled internally by the OS - here, having to hold a non-mutable reference to a particular IP stack seemed like a good compromise.

Requiring a `&mut ref` would make operating, say, an HTTP client and an SMTP client at the same time almost impossible.

MathiasKoch, 2020-02-24:
I think that makes sense. I will go with the internal mutability :+1:

closed 2020-02-24 by MathiasKoch

kellerkindt, 2020-12-31:
Cross linking to #39