rust-embedded-community/embedded-nal #6: Async/Await

olanod, 2020-03-30:
Perhaps worth designing those APIs to be async/await friendly from the beginning? Soon everything in the Rust world will be asynchronous and embedded [won't be the exception](https://ferrous-systems.com/blog/async-on-embedded/).

thejpster, 2020-04-07:
Yes, I'm very aware of what Ferrous have been up to. Would you like to propose an API?

ryan-summers, 2020-08-27:
Just to add some bit to the discussion - I don't think it's safe to ever say:

> Soon everything in the Rust world will be asynchronous and embedded won't be the exception.

I disagree. The article you link talks about this as well - async is really nice for cooperative multitasking, but it does _not_ work well for hard real-time systems (e.g. safety critical) because there's no guarantee when other tasks will release their execution. There's always exceptions and specific use cases in software.

I definitely support adding async/await traits, but I don't think they should be the only possible API. Perhaps we can have an `AsyncTcpStack` as well as an `AsyncUdpStack` added that use async read/write functions. Then, drivers that make use of async/await can request the associated async stack.

ryan-summers, 2022-07-25:
In light of that, I'm going to close this issue as resolved - thanks for noting this @chrysn! I had neglected to back-trace through issues after approving the async NAL.

We would welcome any further contributions to the async stack!

closed 2022-07-25 by ryan-summers