rust-embedded-community/embedded-nal #8: Support trait for DNS

MathiasKoch, 2020-04-06:
Would it make sense to add a support trait for DNS lookups?

Somthing along the lines of
```rust
pub trait Dns {
    /// The type returned when we have an error
    type Error: core::fmt::Debug;

    fn resolve(&self, hostname: &str) -> Result<IpAddr, Self::Error>;
}
```

thejpster, 2020-04-07:
This is a good idea! But, perhaps we also need to handle:

* Letting the user choose which record (A, AAAA, TXT, etc)
* Iterating through a set of results (you might get several and need to try them in turn)

I can't recall what the Berkeley socket API looks like, but it's probably awful. How do things like lwip do this?

MathiasKoch, 2020-04-08:
Thats a very good point! I forgot about DNS being a bit more than just a single resolve. 
I will peek at how lwip does this! And probably at the Berkeley socket API as well, just to check.

Will return shortly

MathiasKoch, 2020-04-11:
It seems like the Berkeley socket API is pretty much just `gethostbyname` and `gethostbyaddr`, when it comes to DNS?

Using the following arguments:
- **name** specifies the DNS name of the host.
- **addr** specifies a pointer to a struct in_addr containing the address of the host.
- **len** specifies the length, in bytes, of addr.
- **type** specifies the address family type (e.g., AF_INET) of the host address.

thejpster, 2020-04-11:
Do we want to limit ourselves to platforms that can create an `alloc:Vec`, or should we take an `&mut [DnsResult]` and return a sub-slice of that?

MathiasKoch, 2020-04-11:
I would be a bit sad if we limited it to alloc applications.. I think either the `&mut [DnsResult]` version, or go with one of the heapless alternatives to alloc

MathiasKoch, 2020-04-11:
I am not sure `gethostbyaddr` makes any kind of sense in an embedded context?

So the resulting trait would look very much like (depending on what we go for wrt slice/vec)
```
pub trait Dns {
    /// The type returned when we have an error
    type Error: core::fmt::Debug;

    fn gethostbyaddr(&self, hostname: &str) -> Result<Vec<IpAddr>, Self::Error>;
}
```

Where the `Vec<IpAddr>` might be some kind of `DnsResult` instead, though i am not sure that makes sense in an embedded context either? What would it contain other than address?

thejpster, 2020-04-12:
TXT records, IPv4 A record, IPv6 AAAA record, CNAME, MX, DKIM, SPF - there's lots of DNS records.

MathiasKoch, 2020-04-13:
Yeah, i get that. But neither the Berkeley API, nor rust std::net seems to take that into account? Not in requests, and not in the response.

https://doc.rust-lang.org/1.4.0/std/net/index.html

Nor does what seems to be the most used crate for DNS:
https://docs.rs/dns-lookup/1.0.1/dns_lookup/

PR #9 (2020-06-03): First edition of proposal for dns trait
This PR provides a simple DNS trait, similar to that provided by both `lwip` and the `Berkeley sockets API`, consisting of a simple `gethostbyname` function that resolves a hostname to the first ip address given by `A` or `AAAA` DNS record, depending on the requested `addr_type`, and a `gethostbyaddr` to resolve the hostname given an ip address. The later function has the `addr_type` encoded in the given `IpAddr` input.

This is a proposal for a first edition. It will not cover the full DNS, but it is equivalent of both `lwip` and the `Berkeley sockets A [... cut]
closed 2020-06-03 by thejpster