rust-embedded-community/embedded-nal #19: [Question] what should UdpSocket::read do if no packet has been received?

jonahbron, 2020-11-08:
The description is a little ambiguous.  I'd like some clarification on how `UdpSocket::read` should behave if there is no packet to read.  Should it simply return a packet length of 0, or should it return an Error?

ryan-summers, 2020-11-12:
I would argue that the return value should be `nb::WouldBlock` - there is no error on the UDP port, but there is also no data to be read. Typically, most UDP implementations would wait until a packet is available unless specifically set in non-blocking mode. Using `nb` here should solve our problem, as the user can then `nb::block!()` on the result if they want to wait for a packet.

jonahbron, 2020-11-12:
That's brilliant, perfect.  I'll open a PR to document that behavior.

closed 2020-11-13 by jonahbron