rust-embedded-community/embedded-nal #24: [Question] Clarity on what Mode affects

jonahd-g, 2020-11-12:
I'd like some additional clarity on what Mode does, and how it affects the behavior of both read and write.  In a Write operation, there are 2 possible, and several branching ways of handling it.

1. Write, but device is busy and write operation is not possible
  a. return WouldBlock
  b. block (w/ optional timeout) until device is no-longer busy and write is possible
2. Write can be done and executes
  a. execute command and return early, assuming write will be successful
  b. block (w/ optional timeout) until write is complete and successful

For Read, there are more scenarios

1. Read, but no packets have been received
  a. return WouldBlock
  b. block (w/ optional timeout) until a packet arrives
2. Read and a packet is available
  a. read the packet into the buffer

Which of these decisions are fixed in the spec, and which are controlled by the Mode enum?  Note that there is no meaningful concept of "returning early" for the read operation.

ryan-summers, 2020-11-12:
It occurs to me that with the `nb::Result` assigned to the return values, the `Blocking` vs `NonBlocking` modes may largely be meaningless. We could just have all operations be non-blocking and allow the user to selectively block using `nb::block!()` calls. Additionally, `Timeout` is likely too complex for most embedded applications to implement.

Are there some use cases for specific `Mode` values in general?

jonahd-g, 2020-11-12:
I agree, we should lean on `nb` here instead. And yes, getting Timeout to work would definitely be a pain and should happen outside of the driver.

The only use-case that I like is the ability to `write` without waiting for completion.  But maybe that could be an argument on the `write` method.

ryan-summers, 2020-11-12:
@MathiasKoch Do you have any thoughts on this?

@thejpster As the original author, do you have any opinions as to how you were expecting `Mode` to be used?

thejpster, 2020-11-12:
My thoughts were simply that: sometimes you read, say, 16 bytes because you need 16 bytes and 15 bytes is no use to you. Sometimes you read 16 bytes because that's the size of your buffer and actually any number of bytes is fine. The third option is to return if the buffer is full or when the timeout expires.

jonahd-g, 2020-11-12:
I wouldn't have thought to provide the ability to read an incomplete packet.  Is that actually useful?

MathiasKoch, 2020-11-12:
TBH I don't really use mode in my implementations.. My implementations are all buffer based, so the read is actually just a copy from one buffer to another, thus the blocking/non-blocking is kind of a moot point.

ryan-summers, 2020-11-13:
> TBH I don't really use mode in my implementations.. My implementations are all buffer based, so the read is actually just a copy from one buffer to another, thus the blocking/non-blocking is kind of a moot point.

This is basically how I've done it as well. I've never found a logical implementation for `Mode` because the underlying network stack is usually inherently blocking/nonblocking by design, so it's not usually configurable (e.g. smoltcp or the w5500)

ryan-summers, 2020-11-13:
> ...
> 
> This would allow you to read a smaller portion of a packet early if you only need part of it, simply based on the size of the buffer. And there would be no need for a Mode.

I think we can expect that this kind of buffering logic would live inside of an implementee of the `embedded-nal`, but shouldn't necessarily be part of the abstract interface, as that would force us to buffer all implementations, which isn't necessarily desirable (we may want zero-copy implementations).

jonahd-g, 2020-11-13:
> I think we can expect that this kind of buffering logic would live inside of an implementee of the embedded-nal, but shouldn't necessarily be part of the abstract interface, as that would force us [...]

I can't tell whether you're agreeing or disagreeing.  If you're just saying that a driver _could_ do that but doesn't necessarily have to, then we're on the same page.

jonahd-g, 2020-11-13:
> This is basically how I've done it as well. I've never found a logical implementation for Mode because the underlying network stack is usually inherently blocking/nonblocking by design, so it's not usually configurable (e.g. smoltcp or the w5500)

Does this mean that `Mode` should go away?  I think it should, personally.

MathiasKoch, 2020-11-14:
I would personally be okay, with it going away

ryan-summers, 2020-11-14:
I'm also on-board with removing the `Mode`.

PR #26 (2020-11-17): Removed Mode enum
/resolves #24
closed 2020-11-17 by eldruin