rust-embedded-community/embedded-nal #25: [Question] Locate remote socket by MAC address

jonahd-g, 2020-11-13:
The W5500 chip allows selecting a remote by either IPv4, or by MAC address.  Any desire or suggestions on how to make embedded-nal support that feature?

ryan-summers, 2020-11-13:
I believe going below the TCP/UDP layer is out-of-scope for the NAL. That is all very stack-specific and isn't clear how the NAL would be appropriate.

I may be reading too deep into your request, but if you're looking for something like DHCP support for the W5500, suggestions over Matrix have recommended using the W5500 in raw mode and implementing a `smoltcp::Device` interface for it and side-stepping the hardware network stack in favor of smoltcp.

jonahd-g, 2020-11-13:
> I believe going below the TCP/UDP layer is out-of-scope for the NAL. That is all very stack-specific and isn't clear how the NAL would be appropriate.
sounds good :+1: 

I may be reading too deep into your request, but if you're looking for something like DHCP support for the W5500, suggestions over Matrix have recommended using the W5500 in raw mode and implementing a smoltcp::Device interface for it and side-stepping the hardware network stack in favor of smoltcp.
I wasn't asking about DHCP, but that is something I'll need to consider at some point.  I'll look into smoltcp, thanks.

closed 2020-11-13 by jonahd-g