rust-embedded-community/embedded-nal #27: Add support for TCP server APIs

ryan-summers, 2020-11-21:
Currently, there is no means to expose a TCP server via the `embedded-nal`. For example, there is no means to `listen()` on a specific port or to `accept()` connections from clients.

Server operations are added for UDP in #21, but that PR does not address the TCP-side of the stack.

One approach would be to refactor TCP similar to UDP:
```rust
pub trait TcpClient {
    fn send();
    fn receive();
    fn connect();
    fn close();
}

pub trait TcpServer: TcpClient {
    fn listen();
    fn bind();
    fn accept();
}
```

This approach would work, but the downside is that the TCP server would be forced to implement the `connect()` function even when it may never be used by the server.

An alternative approach would be:
```rust
pub trait TcpCore {
    fn send();
    fn receive();
    fn close();
}

pub trait TcpClient: TcpCore {
    fn connect();
}

pub trait TcpServer: TcpCore {
    fn listen();
    fn bind();
    fn accept();
}
```

I'm curious to hear @jonahd-g and @MathiasKoch 's opinions on this one

MathiasKoch, 2020-11-22:
I could live with either, but i think it would make sense to get this in before #28, to get some sort of unification of TCP & UDP?

jonahd-g, 2020-11-22:
That's a tough call.  While having a separate Core trait is more conceptually "pure", it does come at the cost of additional complexity.  I'm leaning towards having only two traits.  The UDP traits have the same issue, where the Server necessarily implements a `connect` method that it will not use.  I have a hard time imagining someone wanting to implement the server functionality but _not_ the client.  Plus, they could implement connect as a noop pretty easily.

ryan-summers, 2020-11-23:
I've been leaning towards @jonahd-g's assessment as well here and opting for the client-server traits as a way to mirror the UDP implementations.

PR #30 (2020-12-02): Adding initial TCP server traits
This PR refactors the TCP traits to align with #21.

This PR fixes #27 by adding a new `TcpServer` trait.

This PR fixes #16 by clarifying the life cycle of a TCP socket. (`socket()` -> `connect()` -> `close()`).

This PR fixes #10 by updating `connect()` to return a `nb::Result` so that connect can be completed asynchronously.

**TODO**:
- [x] Implement a test use case of a TCP server using `std::net` for demonstration purposes.

CC @MathiasKoch @chrysn
closed 2020-12-02 by eldruin