rust-embedded-community/embedded-nal #32: Add an example with TcpStream

fouge, 2020-11-28:
I've been trying to implement the trait for a library I am working on, as a side project. 
I'm struggling to make it work with `std::net::TcpStream`.

My main issue resides in the Error type within TcpClient (I am using the soon-to-be merged branch from Ryan.
The compiler wants to have a `embedded_nal::nb::Error` but this enum needs a type.
if I try to add a type, the compiler consider it to be the returned type and doesn't agree: 

```rust
// with UnixError being a enum deriving Debug.
type Error = embedded_nal::nb::Error<UnixError>;
```

```
error[E0053]: method `connect` has an incompatible type for trait
  --> src/tests.rs:43:2
   |
43 |     fn connect(&self, socket: &mut Self::TcpSocket, remote: SocketAddr) -> Result<(), Self::Error> {
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected enum `embedded_nal::nb::Error`, found enum `tests::UnixError`
   |
   = note: expected fn pointer `fn(&tests::UnixTcpStack, &mut core::option::Option<tests::std::net::TcpStream>, embedded_nal::SocketAddr) -> core::result::Result<_, embedded_nal::nb::Error<embedded_nal::nb::Error<tests::UnixError>>>`
              found fn pointer `fn(&tests::UnixTcpStack, &mut core::option::Option<tests::std::net::TcpStream>, embedded_nal::SocketAddr) -> core::result::Result<_, embedded_nal::nb::Error<tests::UnixError>>`
```

I tried different things with no luck and I am not an experienced Rust developer...

Thanks in advance,
Cyril

MathiasKoch, 2020-11-28:
Hi!

I think your mistake is in the signature of `connect`.

It should return an `nb::Result<(), Self::Error>` rather than a `Result<(), Self::Error>` as you are attempting here.

Your fix should be as simple as:
```rust
type Error = UnixError;
...
fn connect(&self, socket: &mut Self::TcpSocket, remote: SocketAddr) -> nb::Result<(), Self::Error>
...
```

fouge, 2020-11-28:
Hi @MathiasKoch and thank you for the quick answer.

Turns out the signature in the Trait is not correct as well for `socket()` (previously `open()`), `is_connected()` and `close()`.
I changed the signatures in `embedded-nal` and it did the trick.

MathiasKoch, 2020-11-28:
I think only connect, send and receive should be nb:: Result

fouge, 2020-11-28:
Right, I just realized that it was totally ok. 
Thanks Mathias, 
Closing the issue :)

closed 2020-11-28 by fouge