rust-embedded-community/embedded-nal #38: Socket as reference when closing

fouge, 2020-12-09:
Hello,
I was wondering if there wasn't an error when closing the socket

Shouldn't the signature be:

```rust
	fn close(&self, socket: &Self::TcpSocket) -> Result<(), Self::Error>;
```

Instead of:

```rust
	fn close(&self, socket: Self::TcpSocket) -> Result<(), Self::Error>;
```

So that it can take the reference of the socket.

Best,
Cyril

ryan-summers, 2020-12-09:
The intent is that the socket is no longer usable once it has been closed, which is why the close function is intended to take ownership of the socket again. The idea is that you are given a socket in `socket()` and you are returning the socket when you call `close()`.

fouge, 2020-12-09:
Hey @ryan-summers 

I have to say I am a bit confused as my compiler doesn't accept to move the value. Rust can be really frustrating sometimes... 

The only choice I have left looks like I need to `mem::replace` the value somehow...
https://doc.rust-lang.org/error-index.html#E0507

:thinking:

caemor, 2020-12-09:
The 507 error tells you that there should be an open borrow of your socket somewhere most likely. Do you have a code snippet (or the full repo) where the error is happening?

fouge, 2020-12-09:
Here is a snippet:

```
pub struct Rpc<'a, S> {
	socket: S,
	ws: ws::WebSocketClient<SmallRng>,
	in_buf: [u8; 4096],
	out_buf: [u8; 4096],
	tcp: &'a dyn TcpClientStack<TcpSocket=S, Error=TcpError>,
	cmd_id: usize,
}
```

```rust
	pub fn disconnect(&mut self) -> Result<(), RpcError> {
		// initiate a close handshake
		let send_size = self.ws.close(WebSocketCloseStatusCode::NormalClosure, None, &mut self.out_buf)?;
		self.tcp.send(&mut self.socket, &self.out_buf[..send_size])?;

		// read the reply from the server to complete the close handshake
		let received_size = self.tcp.receive(&mut self.socket, &mut self.in_buf)?;
		let ws_result = self.ws.read(&self.in_buf[..received_size], &mut self.out_buf)?;
		match ws_result.message_type {
			WebSocketReceiveMessageType::CloseCompleted => {
				// we can close the TCP socket as well
				self.tcp.close(self.socket)?;
				Ok(())
			}
			_ => {
				Err(RpcError::ErrorClosing)
			}
		}
	}
```

ryan-summers, 2020-12-09:
You can't have the socket be a member of your structure, since the structure **always** owns it and it's not possible to move. Instead, you should store your `socket as an `Option<S>` and move it in/out of the `Option` when you open/close it.

fouge, 2020-12-09:
Thank you guys!
That kind of obvious that my structure shouldn't store something that can be destroyed... :thinking: 

It works now. :pray: 
Cyril

closed 2020-12-09 by fouge