rust-embedded-community/embedded-nal #65: Proposal: embedded-nal-async

lulf, 2022-05-02:
I'd like to propose following a similar path to embedded-hal-async and add an embedded-nal-async crate within this repo. I have been maintaining some async traits for TCP[^1] and DNS[^2] that I feel would be a good fit for this crate (assuming some adjustments to fit the existing embedded-nal conventions). I'm available to take on that effort.

[^1]: https://github.com/drogue-iot/drogue-device/blob/main/device/src/traits/tcp.rs
[^2]: https://github.com/drogue-iot/drogue-device/blob/main/device/src/traits/dns.rs

ryan-summers, 2022-05-02:
I see no reason not to support that here :) It indeed matches the approach of embedded-hal-async.

MathiasKoch, 2022-05-02:
I think that makes great sense! :+1:

PR #66 (2022-05-04): Add embedded-nal-async
* Add embedded-nal-async crate containing async traits requiring Generic
  Associated Types (GAT) feature on nightly for now.
* Add async traits for TCP, UDP and DNS.
* Add embedded-nal-async build to CI

Example implementations:

https://github.com/drogue-iot/drogue-device/blob/embedded-nal-traits/device/src/drivers/wifi/esp8266/mod.rs#L419
https://github.com/drogue-iot/drogue-device/blob/embedded-nal-traits/device/src/drivers/tcp/std.rs#L28
https://github.com/drogue-iot/drogue-device/blob/embedded-nal-traits/device/src/drivers/dns.rs#L34

Fixes #65
closed 2022-05-04 by eldruin