rust-embedded-community/embedded-nal #84: New release of `embedded-nal`

elpiel, 2023-05-03:
It seems that there is small discrepancy between the embedded-nal & embedded-nal-async crates which causes some issues for me. embedded-nal current release has dependency on `no-std-net 0.5` but `-async` has on `0.5`.

Even though `no-std-net` hasn't had any release in a while and is not actively maintained, there is a small difference and with the new support for `ip_in_core` it seems fitting to make a new release of `embedded-nal`.

ryan-summers, 2023-05-10:
I'm not sure I fully understand what the issue you're referring to is? You mention that both use `no-std-net` at 0.5, but is it that one is 0,4 and the other is 0.5?

elpiel, 2023-05-10:
@ryan-summers yes, I had a typo in my issue sorry about that.
So yes, because one version is different than the other I was getting issues with my dependencies.

Quick note that I reached out to the `no-std-net` maintainer and we're discussing the additional of new maintainer(s) to cut out a new release of the crate too.

PR #87 (2023-06-22): Preparing for a 0.7.0 release
Releasing 0.7.0 of the `embedded-nal` to address some TCP soundness issues.

Fixes #84 

I'd also like to take a look at revamping the TCP API to be more like `embassy-net`, but I'm unsure if we should block the release of a new version on potential future work for now.

Thoughts @rust-embedded-community/embedded-nal ?
closed 2023-06-22 by eldruin